nliaudat/esp32_8ch_motor_shield #2: Request for clarification

nbostrom, 2021-05-08:
**Describe the problem you have/What new integration you would like**

First of all, nice work! I have been looking for something like this since we bought our new house. Off the shelf products are unreasonable expensive. I'm a programmer, but I don't have much experience in electronics and embedded software, tho I do have some soldering experience. I'm sorry if I'm not supposed to use the issue board for these kind of questions, but I couldn't find any other contact information. So I guess you that you can see this feature request as a request for some clarification in the docs, if you like.

- Do you have any specific recommendations around the parameters when ordering assembled boards from JLPCB? I have never done anything similar before. 
- I have 13 valves to control. In the readme you recommend getting two boards. Do you mean two completely separate boards with one ESP32 per board?

Thanks again. I hope that I can contribute with a DIN-compatible box with seperate holes for the contacts and maybe help iterate on the software part.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nliaudat, 2021-05-08:
Hi, 
For JLPCB, no special parameters, 2 layers are mandatory.

To control 13 valves, it depends on your programmation and valves : Esp32 has limited ACD sensor available if wifi enable (look at https://github.com/nliaudat/esp32_8ch_motor_shield/blob/main/PCB/Schematic.pdf)

Actually, 2 valves are on the same ADC sensor and do not run at the same time to have a precise BEMF (endstops). The pcb has 2 free ADC sensors available for a future extension board. In your case, I recommend 2 separate pcb with an Esp32 per board.

If you want to spare 4$ for the second ESP32, you can control the second board with the primary board. You'll have limited ADC sensors but it would work as expected.

I have that board in production for 1 year controlled by hassio, with homematic valves. That kink of valves are not very strong and I've made my own as replacement. The why I've made that pcb is I cannot control the main heating temperature (set for all appartments). The previous solution are on/off only and I wanted to have the solution to open the valve 10% only.

If you can set the main water temperature, that pcb is useless.

The future improvement will be a real webserver to set temperatures. With Esphome, the webserver is for debug purpose only. I haven't done it cause set the temperatures with hassio.

Regards.

nbostrom, 2021-05-08:
Hi,
Thanks for the clarifications. Much appreciated!

I can indeed control the main heating temperature heating curve, based on the outside temperature. But my problem is balancing, which becomes very cumbersome when you have 13 valves. Being able to partially open the valves is gold. I hope to be able to automatically balance the valves through software.

I do not think that the extra delay that comes with only running one actuator at a time will be a problem. Our heating tubes are  mold into the concrete and the heating delay is very large either way.

Do you mean that you created your own valve actuators? If so, that's very cool. What problems did you encounter with the homematic actuators? Did they break?

nliaudat, 2021-05-09:
[image]

some Homematic valve do not support the pressure and "unclip"

nbostrom, 2021-05-09:
I see. Thanks for the heads up!

closed 2021-05-14 by nbostrom