nliaudat/esp32_8ch_motor_shield #4: [FEATURE REQUEST] add temp. sensor for VL and RL per channel for doing thermal balancing?

linuxdep, 2021-11-08:
**Describe the problem you have/What new integration you would like**

For thermal balancing i think it was nice to measure VL and RL and calculate thermal balancing automatically.
What do you think is this possible? 
That would be the killer application, install underfloor heating, control it, thermal balancing is done automatically. Suspension of control for the valves marked as closed.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nliaudat, 2021-11-08:
Please give a definition of your mean of VL and RL (German abbreviation ?)
At best a calculation formula.

Actually, the target position of valve is calculated with a simplification of arduino map function : 
```
arduino map function :  return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min; 
with parameters : in_min=-1;in_max=1;out_min=0;out_max=1
=>  f(x) = (x+1)/2
diff_temp = (target_temp_high + target_temp_low)/2 - current_temp
=> round(((diff_temp+1)/2)/0.1)*0.1;
```
I use that formula for 2 years and it works great, is easy to understand and fast to compute. 

For temperature fine tuning, I've tested [PID climate controller](https://en.wikipedia.org/wiki/PID_controller#Integral_term), but my low temperature floor heating has too much inertia to find good values for ki, kp ,kd

The next step I'll go is to fine tune the target valve values with hassio automations. It would be most easy cause hassio retains historical values and can have more sensors, like external temperature, etc.

You can get temperature from external sensors like [Xiaomi BLE](https://esphome.io/components/sensor/xiaomi_ble.html?highlight=xiaomi_lywsdcgq) or via hassio and then compute a [template sensor](https://esphome.io/components/sensor/template.html?highlight=template)

Karl-opec, 2021-11-08:
Hello, great project,

VL means supply flow,
RL means return flow.

I have on my actual Floor Heating system (standard wax based Actuators to control flow) on top of the PWM a PID controller running. Its  PID_PFactor is 0.95, PID_IFactor is 0,59, PID_DFactor =0.5, The D-Value is based on a gradient t(n) to t(n-7) with recalc time of the PID controller of every 60 sec. It works good.  My experience is that the PID controller is needed to keep the Temperature  close to the desired Temp.

nliaudat, 2021-11-08:
They are 7 free pin available to add custom sensors

[image]

Look at https
[image]

Look at [schematics.pdf](https://github.com/nliaudat/esp32_8ch_motor_shield/blob/main/PCB/Schematic.pdf) for details.

You can use [NTC sensors](https://esphome.io/components/sensor/ntc.html?highlight=ntc) or values from hassio. 

Code need to be rewritten a bit (THx_check), but it's simple task

nliaudat, 2021-11-09:
The new firmware version lets you change the valve position formula (at line 53). You can adapt it to get what you want. 
Regards
(https://github.com/nliaudat/esp32_8ch_motor_shield/blob/main/firmware/esphome_firmware_sample.yaml)

nstadigs, 2021-11-10:
Maybe you already know, but apparently esphome has a [PID Climate component](https://esphome.io/components/climate/pid.html) with [support for autotuning](https://esphome.io/components/climate/pid.html#autotuning).

nliaudat, 2021-11-10:
Yes, you can replace thermostat component with PID Climate, I works the same and a sample code is provided in the yaml file (commented out)

nliaudat, 2021-11-10:
Sorry, the code was not fully included : 

`climate:
  - platform: pid
    id: office_pid_thermostat
    name: "Office PID climate"
    sensor: Office_temperature
    default_target_temperature: 21°C
    heat_output: output_office_cover
    control_parameters:
      kp: 18
      ki: 2300
      kd: 0.5

output:
  - platform: template
    id: output_office_cover
    type: float
    write_action:
      - cover.control:
            id: CH6_cover
            position: !lambda return state;
      - logger.log:
            format: "CH6 moved to position %.1f"
            args: [ 'state' ]       `

I have test the climate component, but the update frequency is to hight for valve good life

heinemannj, 2023-09-29:
Is it also possible to use 12 x 1-wire temperature sensors (DS18B20) connected to your board?

I've already implemented via espHome with
[Smart Thermostat with auto Heat/Cool mode, PID control and PWM (Home Assistant component)](https://github.com/hacker-cb/hassio-component-smart-thermostat#smart-thermostat-with-auto-heatcool-mode-pid-control-and-pwm-home-assistant-component) and this working fine with my actual ON/OFF Actuators.

But I want to change to Homematic valve actuators [HmIP-VDMOT] for some reasons ;-)

nliaudat, 2023-09-29:
Yes, you only need a VCC, GND, IO. 
All is available in https://github.com/nliaudat/esp32_8ch_motor_shield/blob/main/extension.md

heinemannj, 2023-10-02:
Coming back to the origional topic: **thermal balancing**

In Germany, a hydraulic balancing must be carried out after installation or changes to a heating system.

However, if the heating circuit conditions like line cross-sections and line lengths of the individual heating circuits are not known, an exact calculation of the required flow rate is not possible.

For this reason, only a thermal calibration can be carried out.

By measuring the supply flow temperature and the return flow temperature in each individual heating circuit, you can calculate the spread.

Ideally the spread in each heating circuit within one thermal zone (e.g. room) should be equal.

For the thermal calibration you have three possibilities:

1. Adjust the flow via your flow meters
2. Adjust the flow via proportional valve actuators
3. Adjust initially via flow meters and then daily on demand based on room and outside temperature conditions via proportional valve actuators

The third is obviously the best method because then you have setup a real `dynamic hydraulic balancing.`

As mentioned by [Karl-opec](https://github.com/Karl-opec) an external HA PID controller can manage this (please see above).

It is also mentioned in [floor-heating-controller](https://github.com/nliaudat/floor-heating-controller) wiki that you can use Hassio custom climate components.

Can you please provide some example code for the usage/integration of Hassio components to control the valve percentage of the actuators.

nliaudat, 2023-10-02:
There is 2 ways to go to do external PID control : 

The simplest is to adjust the target temperature (like in https://github.com/KartoffelToby/better_thermostat)
The second is to control valves motors and then use the "motor firmware", but I have no experiences in that way.

nliaudat, 2023-10-02:
https://github.com/vindaalex/multizone-thermostat can control proportionnal valves

heinemannj, 2023-10-13:
I've setup an very early alpha version of a documentation / project paper to describe the goals and the physics behind:

https://github.com/heinemannj/Hydraulic-Balancing/edit/main/README.md

closed 2024-02-24 by nliaudat