nliaudat/esp32_8ch_motor_shield #5: Move the rj11 connectors a little bit over the edge of the board

nstadigs, 2021-11-22:
It would be great if the connectors stuck out from the board by a couple of millimeters.

**Please describe your use case for this integration and alternatives you've tried:**

I wanted to create a custom 3d printed box for my two boards with individual holes for the separate rj11 connectors, but it's hard to do when they are aligned with the edge of the board.

**Additional context**

nliaudat, 2021-11-22:
[image]

There is a track for connecting motors without an RJ45 connector which is at the edge of the board and which locks the location of the RJ45 connector. I'll do some tests to move it, but in the meantime the easiest way is to cut the board.

nstadigs, 2021-11-22:
Oh, that's great to hear! Then I'll just cut it :)

Thanks!

nliaudat, 2021-11-23:
done in v53 release

closed 2021-11-23 by nliaudat