nliaudat/esp32_8ch_motor_shield #6: ESP hangs and L9110S gets very warm

nstadigs, 2021-11-22:
Hi!

I understand if this isn't a support channel for newbies like me, but I'm desperate so I'll give it a try anyhow.

I ordered a couple of boards from jlcpcb with chips assembled. The only change I made was selecting [this L9110S](https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C85174) instead of the [one in the BOM](https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C85174), because that one was out of stock for several months. They seemed compatible, but again, I'm new to this.

I did not have any issues soldering the through hole components and I have tested all the joints.

The issue: The L9110S runs warm and the motors are supplied constant voltage of 1.3 volts, even when there is no ESP connected to the board.

Did I select an incompatible L9110S? Do you have any troubleshooting tips :grimacing: 

Thanks :pray: 

PS: If you were to sell ready-made boards I'd buy two of them ;)

nliaudat, 2021-11-22:
Which motor do you use ? 
Have you a 3.3V power source and a multimeter to make tests ?
The only difference I see with the original chip is the max voltage supported 12V for mine and 7V for yours
I mean a 10K resitor is broken or not good soldered

With the EPS not connected to the shied, are the motor running on each 8 channels ?

nstadigs, 2021-11-22:
> Which motor do you use ?

HmIP-VDMOT

> Have you a 3.3V power source...

I have and HLK-5M03 soldered to the board and the backside of the board secured.

> ...and a multimeter to make tests ?

Sure do!

> The only difference I see with the original chip is the max voltage supported 12V for mine and 7V for yours

Sounds promising, thanks

> I mean a 10K resitor is broken or not good soldered

Ok..
 
> With the EPS not connected to the shied, are the motor running on each 8 channels ?

Yes, its running on each channel

nliaudat, 2021-11-22:
I'll do some tests this evening, but the BEMF (back electromotive force) should be 0 cause no motor is connected.
Is the ESP connected or not ? Perhaps it's a faulty SN74HC595 shift register
Can you put a picture of the front and the back of your board ? I'll check the soldering

nliaudat, 2021-11-22:
Are the L9110S chips turned correctly in the right direction ?
[image]

nstadigs, 2021-11-22:
The esp isn't connected.

The the point on the L9110S is next to the point on the board. Lower left .

Thanks! I'll send you some pictures later

nstadigs, 2021-11-22:
[image]
[image]
[image]
[image]
[image]

nliaudat, 2021-11-22:
Have you jumper cables ?

Can you check the motor voltage if you link IA to GND and IB to VCC, and after that IA to VCC and IB to GND.
Values should be + or - 3.3V
Test only on channel 1
(they are many holes labeld VCC and GND on the board)

nliaudat, 2021-11-22:
Some of your soldering are very bad. 
[image]

All soldering must be disconnected, chrome and brilliant (not dull nor yellowed). Sorry for my bad english :(

nliaudat, 2021-11-22:
[image]

nstadigs, 2021-11-22:
2.2 volts and 1.7 volts

Yeah, I figured my soldering wasn't the best. But I thought it would be good enough :)
I'll try to fix it to see if that fixes the problem.

Thanks again!

nliaudat, 2021-11-22:
I've reproduced your problem inverting the L9110s module upside down. I've got the same results. 

I can send you 2x v1.0 prototype boards and you send me back 2 of your to do more checks if you want.

closed 2021-11-22 by nliaudat