nliaudat/esp32_8ch_motor_shield #7: [FEATURE REQUEST] Use an external power adapter?

nstadigs, 2021-11-24:
**Describe the problem you have/What new integration you would like**

If you would add support for using an external power adapter then the users would not have to deal with mains power. I do think that installing this legally in some countries would actually require a certified electrician if that's even allowed without CE mark.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nstadigs, 2021-11-24:
Sorry, I just realized that you fixed this in 1.4 :man_facepalming:

closed 2021-11-24 by nstadigs

nliaudat, 2021-11-24:
You can already use some external power adapter like [that one](https://fr.aliexpress.com/item/32968108837.html?spm=a2g0o.productlist.0.0.44302fb8o4JDbt&algo_pvid=06e25e76-df35-46a2-a341-723292da4e8d&algo_exp_id=06e25e76-df35-46a2-a341-723292da4e8d-1&pdp_ext_f=%7B%22sku_id%22%3A%2210000015054998152%22%7D). 
Everything with 3.3V and 1.25 A (~ 4 Watt) can do the job.
The price is ~3x greater than HLK-5M03. 
You have to link it with any of the VCC-GND pin labeled on the board.

nstadigs, 2021-11-24:
You're totally right. I was posting before thinking. I was thinking about an actual connector on the board. But I could just make a hole in the box for a [connector like this](https://www.amazon.se/dp/B07N7G7MCP/ref=sr_1_3?keywords=dc+honkontakt&qid=1637741353&sr=8-3) and connect my two boards to that connector in parallel on the inside.

nliaudat, 2021-11-24:
Usually, there is no power source in the valve manifolds location. You must disconnect an existing 230V valve and wire it to the board. That's the why the external power was not a solution. 

You can use a 230V power socket like that
[image]

nstadigs, 2021-11-24:
Interesting. I have a standard EU plug like this so I though it was common. I know other houses around here have the same solution. The Uponor controller, which most people pick around here has a standard plug for it:

[image]