nliaudat/esp32_8ch_motor_shield #8: In the latest version + and - are connected

nstadigs, 2021-12-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

N/A

**ESP (ESP32/ESP8266, Board/Sonoff):**

N/A

**ESPHome version (latest production, beta, dev branch)**

N/A
**Affected component:**

N/A

**Description of problem:**
In the latest version 3.3v and ground are connected

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

It's probably this, right?

[image]

nliaudat, 2021-12-16:
I see. 
Can you send the part number of the HLK 3.3V power supply.

nstadigs, 2021-12-16:
It's the HLK-5M03 provided in the BOM file: C209906

nstadigs, 2021-12-16:
To clarify, they are connected even on a bare board with nothing soldered

nstadigs, 2021-12-16:
I should be able to just drill out these VCC holes to see if that fixes the problem, right? If so, I can just connect the 3.3v elsewhere with a cable. What do you think?

nstadigs, 2021-12-16:
I can confirm that this solved the issue
[image]

nliaudat, 2021-12-30:
Sorry for that issue. The HLK footprints were wrong. I'll add extra space in next release

nliaudat, 2021-12-30:
Fixed in release V54

closed 2021-12-30 by nliaudat