nliaudat/esp32_8ch_motor_shield #9: Unable to read BEMF

nstadigs, 2021-12-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-DevKitC-v4
**ESPHome version (latest production, beta, dev branch)**

ESPHome version 2021.12.1
**Affected component:**

adc

**Description of problem:**
ADC sensor senses nothing even tho I get small values when measured at the pins.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: floor-heating-controller-1
  platform: ESP32
  board: nodemcu-32s

wifi:
  networks:
  - ssid: !secret wifi_ssid_1
    hidden: true
    password: !secret wifi_password_1
    priority : 1

mqtt:
  broker: 192.168.0.186

ota:

web_server:
  port: 80

sn74hc595:
  - id: 'sn74hc595_hub'
    data_pin: GPIO16 #Pin connected to SN74HC595 SER input
    clock_pin: GPIO5 #Pin connected to SN74HC595 SRCLK pin
    latch_pin: GPIO17 #Pin connected to SN74HC595 RCLK pin
    oe_pin: GPIO18 ### needed by esphome, but connected to GND (outputEnable is used only for pwm all th pin)
    sr_count: 2 #Number of daisy-chained shift registers, up-to 4

# Enable logging
logger:
  level: DEBUG #NONE# ERROR #INFO #DEBUG  #VERBOSE
  baud_rate: 0 #to disable logging via UART
  logs:
    adc: INFO

switch:
    ### CH1
  - platform: gpio
    name: "CH1 IA"
    pin:
      sn74hc595: sn74hc595_hub
      number: 0
      inverted: False
    internal: false
    id: CH1_IA_pin
    interlock: &interlock_group_CH1 [CH1_IA_pin, CH1_IB_pin]
    restore_mode: always off
  - platform: gpio
    name: "CH1 IB"
    pin:
      sn74hc595: sn74hc595_hub
      number: 1
      inverted: False
    internal: false
    id: CH1_IB_pin
    interlock: *interlock_group_CH1
    restore_mode: always off

sensor:
  - platform: adc
    id : BEMF_1_2_sensor_ADC
    pin: GPIO36
    name: "BEMF 1" #CH1 + CH2
    update_interval: 500ms
    attenuation : 2.5db

```

**Logs (if applicable):**

```
[09:24:25][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:25][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:26][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:26][D][switch:021]: 'CH1 IA' Toggling ON.
[09:24:26][D][switch:037]: 'CH1 IA': Sending state ON
[09:24:26][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:27][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:27][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:28][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:28][D][switch:021]: 'CH1 IA' Toggling OFF.
[09:24:28][D][switch:037]: 'CH1 IA': Sending state OFF
[09:24:28][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:29][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:29][D][switch:021]: 'CH1 IB' Toggling ON.
[09:24:29][D][switch:037]: 'CH1 IB': Sending state ON
[09:24:29][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:30][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:30][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:31][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:31][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:31][D][switch:021]: 'CH1 IB' Toggling OFF.
[09:24:31][D][switch:037]: 'CH1 IB': Sending state OFF
[09:24:32][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:32][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:33][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[09:24:33][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
```

**Additional information and things you've tried:**

Board: v53 rev 1.19 

Putting a multimeter between GPIO36 and GND gives around 20mV when the motor is running unhindered and around 50mV at end stops. Are these values normal?
If I set the ADC sensor to report "raw" I get only 0's. As I understand I should get a value between 0 and 4096.
Maybe these values are too low to be measured by my esp32? Or are the values from my board lower than expected?

I've tried other pins and motor contacts but with the same results.

PS. I haven't received the old board you sent me yet, but I payed the import taxes for them yesterday. The local postal service had entered the wrong address for the payment notice letter so it was brought here by my neighbor.  So they will probably arrive soon. Thanks again :)

nliaudat, 2021-12-17:
Please try to set `attenuation: auto` 
[ADC sensor](https://esphome.io/components/sensor/adc.html) change in 2021.11 esphome release. 
My Esp32 works great with 2021.11 release but it depends on manufacturer calibration

nstadigs, 2021-12-17:
Thanks, I've tried that. That seems to set it to 2.5db automatically, giving the same values.:
```
[10:01:45][D][sensor:113]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[10:01:45][D][switch:021]: 'CH1 IB' Toggling ON.
[10:01:45][D][switch:037]: 'CH1 IB': Sending state ON
[10:01:45][D][sensor:113]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[10:01:46][D][sensor:113]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[10:01:46][D][sensor:113]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[10:01:46][D][switch:021]: 'CH1 IB' Toggling OFF.
[10:01:46][D][switch:037]: 'CH1 IB': Sending state OFF
[10:01:47][D][sensor:113]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[10:01:47][D][sensor:113]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
```

nliaudat, 2021-12-17:
ADC values are proportional to vref (esp32 input voltage). Can you check input voltage ?

nstadigs, 2021-12-17:
Input is 3.30 V

nliaudat, 2021-12-17:
Your voltage reference between GPIO36 and GND are correct values. That means it's an ESP32 problem.
Can you test : 
```
raw: true
filters:
  - multiply: 0.00026862
```
and give details or picture of the ESP32 Chip Surface Marking

nstadigs, 2021-12-17:
```
[10:58:50][D][sensor:113]: 'BEMF 1': Sending state 0.00000 V with 2 decimals of accuracy
```

nstadigs, 2021-12-17:
[image]

nliaudat, 2021-12-17:
What is the chip date code ?
[image]

nliaudat, 2021-12-17:
Can you test to wire VCC 3.3V to GPIO36 or better put a potentiometer between ?
[image]

nliaudat, 2021-12-17:
Hi, I've done some tests on my board between esphome 2021.10.3 and latest. The values of ADC sensors are 2x biggers. 
With my ESP32 devkit, I can measure ADC at 0.04V. 
Your board is perhaps less sensitive. 
You can try to cut the 12ohm resistor with a with wire cutters. Watch out for projections in your eyes !

[image]

nstadigs, 2021-12-17:
I can't find the date code anywhere on my chip :thinking: 

I don't have a potentiometer but I tried with a bunch of different resistors and I can actually read some higher values via ADC!

You are probably right about the sensitivity of my board. I will try to cut a resistor tonight.

Thanks, you are very helpful!

nstadigs, 2021-12-17:
I removed that 12ohm resistor and now I sometimes get a little spike right when the motor starts, but then it returns to "normal". I doesn't go up and end stops either :man_shrugging:

```
[19:19:30][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[19:19:30][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[19:19:31][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[19:19:31][D][switch:021]: 'CH1 IA' Toggling ON.
[19:19:31][D][switch:037]: 'CH1 IA': Sending state ON
[19:19:31][D][sensor:113]: 'BEMF 1': Sending state 0.09400 V with 2 decimals of accuracy
[19:19:32][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[19:19:32][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[19:19:33][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[19:19:33][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[19:19:34][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[19:19:34][D][switch:021]: 'CH1 IA' Toggling OFF.
[19:19:34][D][switch:037]: 'CH1 IA': Sending state OFF
[19:19:34][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[19:19:35][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
[19:19:35][D][sensor:113]: 'BEMF 1': Sending state 0.07800 V with 2 decimals of accuracy
```

nstadigs, 2021-12-17:
The values from the multimeter is now 19mV when running and 50~70mV at end stops

nliaudat, 2021-12-18:
You can replace the "smoothing" resistor if you have one with a value less than 12ohm between BEMF and GND.
[image]

To adjust the endstop threshold, there are bemf_triggers 1 to 8 (input_numbers)

nstadigs, 2021-12-18:
Got myself a breadboard, some jumper cables and a potentiometer today to really dig in.

Seems like the smallest values I can measure via ADC is around 100mV. Everything below that comes out as 0. This matches the recommended measuring values on [the docs on the espressif website](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html#_CPPv425adc1_config_channel_atten14adc1_channel_t11adc_atten_t). Even then it fluctuates a lot until I get to around 200-300mV. Is there a good way to amplify the values from the board?

nstadigs, 2021-12-18:
My bad! I did not understand that the resistors of CH1 and CH2 where in parallel! After removing the other 12ohm I can successfully read values! But it seems that I have to replace the "smoothing" resistor with one more that 12ohm rather than one with less to read values. I guess a lower value one practically "dumps" more voltage into GND? Anyhow. I now have 22ohm resistors connected as you described and it seems to work!

nliaudat, 2021-12-19:
I'll remove that resistors in the next version. They make troubles

nstadigs, 2021-12-19:
Ok! Sorry, I don't understand half of this. What are they there for? It seems like they make the values less "jumpy". Is this correct?

Do you also get a spike when the motor starts sometimes? That spike is often higher than the endstop bemf causing the motor to stop immediately. Is this the reason why you have the small 5 second pause in the start of the calibration script?

nliaudat, 2021-12-19:
I've just done some test and they are necessary. Without them, the GND is not connected to the L9110 module. If linked directly to GND without resistor, no way to read BEMF

nliaudat, 2021-12-19:
For the spikes, you can adjust the BEMF_trigger_x value. 
I have only small spikes about 15 mv and then running at 8-10mv

nstadigs, 2021-12-19:
Yeah I noticed that as well. Without the resistor s the bemf goes bananas. Regarding the spike it's often higher than the BEMF at end stops for me, but I think that I can solve that in the software

nstadigs, 2021-12-19:
18ohm resistors seems to be optimal for me. If that works for you too maybe it's best to replace them in a future version so that other won't run in to the same problems?

nliaudat, 2021-12-20:
Actually there are two 12ohm parallel resistors = 6ohm
If you remove the parallel resistors and link BEMF with GND, the resistors are in series and add their values (18+18=36ohm)

Strange it doesn't work with the factory 6ohm resistors....

nstadigs, 2021-12-20:
Oh my. The resistors on my board are wrong! They are marked 1R07! Those are 1.7ohm resistors, right? Did jlcpcb mount the wrong resistors?

nstadigs, 2021-12-20:
When I mount 12ohm resistors it works

closed 2021-12-30 by nliaudat

lal12, 2022-02-13:
According to the schematic and BOM a 1.07Ω shunt is used. According to homematic IP the motors are using 120mA of current. This would mean a drop of of 128mV over the shunt. And according to the ESP32 docs the minimum voltage the ADC can measure is 100mV. This seems to be a bit tight especially considering the 120mA are maximum ratings and we are using a bit of the 3.3V for the shunt so the real current would at least a bit lower. 

Just found [this](https://github.com/Lenti84/VdMot_Controller/blob/developer/system/actuators.md) where they measured a current of ~15mA for this motor. So that would only be 16mV far too less for the ESP to measure.

nliaudat, 2022-02-13:
Many thanks for the comment. I did not knew the https://github.com/Lenti84/VdMot_Controller project. 
I work on a complete refractor of the project with custom valves as the homematic Vmot are bad quality (they jump around their support and breaks). 
What is your recommendation for the resistors value ? I've made many tests and the best is 12kohm in series but L9110s chip is not well documented and interfering

nstadigs, 2022-02-14:
> Many thanks for the comment. I did not knew the https://github.com/Lenti84/VdMot_Controller project. I work on a complete refractor of the project with custom valves as the homematic Vmot are bad [...]

I too have been thinking about custom valve motors for some time. I started experimenting with a custom ESPhome component for your board that counts motor revelations with code borrowed from https://github.com/Lenti84/VdMot_Controller. But the values from the BEMF relies on the amount of pressure caused by your specific valve pins, so that gets pretty complex. So instead I started to look into using stepper motors for driving the valves. I found these [small 5v stepper motors with integrated reducing gears and a 3m thread in the pin](https://www.aliexpress.com/item/32797216294.html?spm=a2g0o.productlist.0.0.49be4fddTDP79d&algo_pvid=58efb47e-3498-4f08-83fe-e16897e50b99&algo_exp_id=58efb47e-3498-4f08-83fe-e16897e50b99-2&pdp_ext_f=%7B%22sku_id%22%3A%2210000001114058835%22%7D&pdp_pi=-1%3B19.52%3B-1%3B-1%40salePrice%3BSEK%3Bsearch-mainSearch) on AliExpress. I'm thinking that if you put an M3 hex head screw in the hole in the pin and lock the rotation of the screw you can use that to press the pin. This would allow fine grained positioning of the valve without constant recalibration. What do you think?

nliaudat, 2022-02-14:
I've made a working valve with https://fr.aliexpress.com/item/4000311295036.html?spm=a2g0s.12269583.0.0.18834b0ejbgKSj
15 RPM, 3V. 

For steppers, a lot a TRV valve manufacturers use https://www.alibaba.com/product-detail/Brand-new-Linear-actuator-for-Valve_60637359967.html?spm=a2700.shop_plgr.41413.12.531f5635mmtYvx

Steppers is a good solution for position handling, but do not survive a controller reboot (lost position value). 

I work on an absolute position control with a reflective optical sensor (ITR8307). That means the controller board must read one more analog input for each valve. The ESP32 is limited in ADC and that involve to use a analog multiplexer (CD74HC4067).

nstadigs, 2022-02-14:
> I started experimenting with a custom ESPhome component for your board that counts motor revelations with code borrowed from https://github.com/Lenti84/VdMot_Controller. But the values from the BEMF relies on the amount of pressure caused by your specific valve pins, so that gets pretty complex.

I was thinking about using a small push button on the top side to determine the top position at startup. This is similar to what 3d-printers do.

> I work on an absolute position control with a reflective optical sensor (ITR8307). That means the controller board must read one more analog input for each valve. The ESP32 is limited in ADC and that involve to use a analog multiplexer (CD74HC4067).

Very interesting! I'm eager to see what you come it with :)

nliaudat, 2022-03-16:
For information, I upgraded today from python 3.7 to 3.10.3 and recompile with latest esphome and now the null value of BEMF sensor is 0.08V instead of 0.0V
Very strange .... I'll investigate if I'll find time to.

dragop4, 2022-12-20:
@nliaudat do you have any solution about this?
I started with this project, build and installed to esp directly from esphome dashboard (Homeassistant). And i have permanent  0.08 V  on BEMF sensor. 
1.35 v54 board version, ESP32-WROOM-32U chip and HmIP-VDMOT actuators.

nliaudat, 2022-12-20:
@dragop4  No solution for 0.08V reading. 
I mean it's related to python 3.8 rounding number changes. 
Reading ADC 0.08 means 0V and that's not a problem to upper the BEMF trigger by 0.1

The latest board as opamp to have better ADC reading.

dragop4, 2022-12-21:
> 

it's possible to modify my board without ordering new?

nliaudat, 2022-12-21:
have a look at https://github.com/nliaudat/esp32_8ch_motor_shield/discussions/19#discussioncomment-3724593