nliaudat/esp32_8ch_motor_shield #10: Order possibilities ?

marcelveldt, 2021-12-30:
Hi! This project looks really interesting and I've been looking for something like this for a while now.
I actually own the Homematic IP actuators etc. but not very happy with it.
Are there any possibilities to order 2 prebuilt actuators/shields ?

nliaudat, 2021-12-30:
Hi, the project is still in early development. No order possibilities actually. Sorry

closed 2021-12-30 by nliaudat

nliaudat, 2021-12-30:
You can order at JLPCB or other PCB manufacturer

marcelveldt, 2021-12-30:
Thanks for the reply!
I'll wait until it is in final shape then.

nethawk2, 2022-01-15:
Hi Nicolas, i would like to order at JLCPCB, they want a gerber file. Can you provide your gerber file for upload to JLCPCB ? 
Tahnk you

nliaudat, 2022-01-15:
All you need is in the release (https://github.com/nliaudat/esp32_8ch_motor_shield/releases)

nethawk2, 2022-01-16:
Thank you. 
One part seems not available for a longer time: JLCPCB Part Nr. C273642, TI SN74HC595PWR
Do you have a suggestion for a replacement ?
Thank you

nstadigs, 2022-01-16:
Hi @nethawk2. That part seems to get in and out of stock frequently. I suggest that you try again in a couple of days. I suspect that this is a result of the global chip shortage.

nethawk2, 2022-01-16:
Thx @nstadigs , on other webpages i saw a schedule of 10 months or even 2023.

I am really not the hardware guy, but i try to learn.
Would Texas Instruments SN74HC595DRG3 , JLCPCB Part Nr. C94270 be a good replacement ?

nliaudat, 2022-01-16:
Not compatible, you must find any SN74HC595 part with the same sock (TSSOP-16). I advise you to wait a week. LCSC has a stock of 2000+

nethawk2, 2022-01-17:
How about 74HC595T16-13 from Diodes Incorporated , Part Nr. C148115 ?
This is my last try to find a substitute to not waste more of your time.

nliaudat, 2022-01-18:
1st : do not write in closed issues.
2nd : that's not an issue
3rd : You can wait for JLCPCB to have stock or find another PCB manufacturer