nliaudat/esp32_8ch_motor_shield #12: Not an issue: I will have 6 assembled boards available for german buyers...

Lonestar78, 2022-09-03:
Dear All,

not exactly an issue ;-)
i just ordered 10 assembled PCBs and can make 6 available for buyers (cheapest shipping in Germany)
Let me know if interessted.

Kind regards
Christian

ChristianBuer, 2022-09-09:
Hi,
I am interested. How much would three of them cost with shipping in Germany (I live in Wuppertal).

Kind regards,
Christian

Lonestar78, 2022-09-10:
Hi, I would say about 22 € per assembled PCB plus DHL. The delivery should be at my door next week.
Kind regards Christian

Holen Sie sich Outlook für Android<https://aka.ms/AAb9ysg>

linuxdep, 2022-09-10:
Hi, schön, ich hätte auch Interesse an 2 PCB's. Ist der ESP32 auch dabei?

Lonestar78, 2022-09-10:
Hi, would be without the ESP32, but i need to check how many i ordered on ali ;-)
If both above tke the PCBs i would have 2 more to offer.
which forum are you in? I can generate an account there.

linuxdep, 2022-09-10:
https://homematic-forum.de/forum/
Da passt es ja hin...

Lonestar78, 2022-09-10:
done... same username as on git

danif86, 2022-09-11:
I need 3, only 1 left? And live in Norway (would pay for shipping)

nliaudat, 2022-09-11:
Hi, I've created a discussion to buy or sell boards : 
https://github.com/nliaudat/esp32_8ch_motor_shield/discussions/categories/buy-or-sell

closed 2022-09-11 by nliaudat

playya, 2022-09-18:
@Lonestar78 I need only a single board and as a software guy to dump to order at JLCPCB.
Is the last board still available?