nliaudat/esp32_8ch_motor_shield #30: [BUG] USB and DC power supply separation by diode

WMP, 2022-12-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**
If a person who will not be familiar with this board/project will think that because it's a USB it can be plugged into the computer, and the board will be powered from DC it will burn the USB in the computer. Could you add protection, such as diode separation?

In addition, the LDO BL1084-33-CY can heat up when the voltage is higher than 9V, can you for the VCC pin give a copper bond to transfer the heat?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

nliaudat, 2022-12-19:
Good idea ! Many thanks for the improvements proposal. 

I have no time to do it actually, and the DC version was never proper tested as nobody use it.
I mean the DC version will disappear soon :(

WMP, 2022-12-19:
I thinks that 230v AC version is not updated, and recently version is usb or DC. I waiting for new version with fix BEMF, and I want order PCB, what version I should order? 230AC or usb/DC?

nliaudat, 2022-12-19:
The AC version PCB v55 rev 1.45 was tested 4 days ago  https://github.com/nliaudat/esp32_8ch_motor_shield/discussions/19#discussioncomment-4412833

closed 2023-03-03 by nliaudat