nliaudat/esp32_8ch_motor_shield #32: Need help

uplight-dev, 2023-01-06:
Could you help me understand how to order this board from https://jlcpcb.com ?
I saw the Release files are not available anymore.

Thanks!

nliaudat, 2023-01-06:
You can order after connection at [easyeda.com](https://u.easyeda.com/join?type=project&key=b2642362910a768d05dcb075a611a4d2&inviter=e63b80a2d0b64d088d67d4a8c7810922). The project is now public

uplight-dev, 2023-01-06:
Thank you!

I tried to order by generating Gerber, BOM, CPL files but at jlcpcb.com I get these errors:
- HLK-5M03 inventory shortage
-  U1 and H1,H3 are missing, cannot be identified; these are provided by Jayesh Prajapati and chan48969999. There's no link in the BOM file for them

Could I go ahead and order without these 3 components and buy them separately and solder them myself?

I found HLK-5M03 here:
https://www.tinytronics.nl/shop/en/power/power-supplies/3.3v/hi-link-pcb-power-supply-3.3vdc-1.5a-hlk-5m03

Thanks!

nliaudat, 2023-01-06:
You can replace the HLK-5M03  with HLK-PM03 (add to bom properties). It's the same with less power. You do not need power if many valves runs at the same time which is not the default config.

U1 is the esp32 dev board, you need to buy it aparts. 
Someone changes it to "add in the Bom". I revert that. (that's why I do not make the project public in early steps)

Regards

uplight-dev, 2023-01-06:
One thing I missed to ask, would this board work with a Raspberry PI with a wired connection?
Or can the antenna be used with Raspberry PI?
I don't want to use a wifi connection with it.

nliaudat, 2023-01-06:
No way, wifi only. 
That's an esp32 board. 

If you have a raspi, why do you want an esp32 shield ? You should find a 8 channel motor driver for rpi. 

You could use the RX/TX port to send command to the board with your raspi, but it's out of support.
Regards

closed 2023-01-06 by nliaudat

uplight-dev, 2023-01-06:
It's because I'd like to use the HmIP-VDMOT actuators.

nliaudat, 2023-01-06:
take a look at https://github.com/Lenti84/VdMot_Controller as alternative.

uplight-dev, 2023-01-06:
While it's out of support, is there  perhaps some similar product or some raspi tutorial on using/connecting the RX/TX ports?
I ask because I think the https://github.com/Lenti84/VdMot_Controller is closed-source and I'm hoping to be able to modify the board myself a bit before printing it.