nliaudat/esp32_8ch_motor_shield #35: [BUG] closed source project on github?

QuadCorei8085, 2023-03-17:
What is the point of the project being on github when no sources are available just images?

nliaudat, 2023-03-18:
The sources where deleted last month since nobody helped nor commited any improvement since 2 years. 
I pushed the easyeda sources and altium designer in [/pcb](https://github.com/nliaudat/esp32_8ch_motor_shield/tree/main/pcb), but not the shared easyeda project on cloud designer. 

Regards

closed 2023-03-18 by nliaudat

WMP, 2023-04-12:
Could you share PCB for version v57 rev 1.48 ?

nliaudat, 2023-04-12:
> Could you share PCB for version v57 rev 1.48 ?

Why do you want an outdated version ? I'll have to search ...

WMP, 2023-04-12:
Because this is last with rj11, I have 10 PCB and I want to make box for this version

nliaudat, 2023-04-12:
The box is on https://github.com/nliaudat/esp32_8ch_motor_shield/tree/main/box

Here you'll find the board PDF for v57 rev 1.48

[PCB_PCB v57 rev 1.48_2023-04-12.pdf](https://github.com/nliaudat/esp32_8ch_motor_shield/files/11214642/PCB_PCB.v57.rev.1.48_2023-04-12.pdf)