nliaudat/esp32_8ch_motor_shield #46: BEMF is NOT working

heinemannj, 2023-10-28:
Valves are working but by reaching the endstop point there is more then 10 secs tock tock tock and the valveis pressed from the ventil.

board version  v57 rev 1.48

[image]

ESPRESSiF
ESP32-WROOM-32D

[image]

Homematic valve actuators [HmIP-VDMOT]

[image]

```
INFO ESPHome 2023.10.3
INFO Reading configuration /config/esphome/floor-heating-1.yaml...
WARNING GPIO5 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from floor-heating-1.local using esphome API
INFO Successfully connected to floor-heating-1.local
[19:13:14][I][app:102]: ESPHome version 2023.10.3 compiled on Oct 27 2023, 17:54:34
[19:13:14][I][app:104]: Project nliaudat.floor-heating-controller version 1.94.dev
[19:13:14][C][logger:416]: Logger:
[19:13:14][C][logger:417]:   Level: DEBUG
[19:13:14][C][logger:418]:   Log Baud Rate: 115200
[19:13:14][C][logger:420]:   Hardware UART: UART0
[19:13:14][C][logger:424]:   Level for 'adc': INFO
[19:13:14][C][logger:424]:   Level for 'dallas.sensor': ERROR
[19:13:14][C][logger:424]:   Level for 'wifi': ERROR
[19:13:14][C][logger:424]:   Level for 'wifi_signal.sensor': ERROR
[19:13:14][C][logger:424]:   Level for 'ota': ERROR
[19:13:14][C][logger:424]:   Level for 'uptime.sensor': ERROR
[19:13:14][C][logger:424]:   Level for 'restart': ERROR
[19:13:14][C][logger:424]:   Level for 'mdns': ERROR
[19:13:14][C][logger:424]:   Level for 'internal_temperature': ERROR
[19:13:14][C][logger:424]:   Level for 'sntp': ERROR
[19:13:14][C][logger:424]:   Level for 'web_server': ERROR
[19:13:14][C][logger:424]:   Level for 'captive_portal': ERROR
[19:13:14][C][logger:424]:   Level for 'api': ERROR
[19:13:14][C][logger:424]:   Level for 'component': ERROR
[19:13:14][C][logger:424]:   Level for 'thermostat.climate': ERROR
[19:13:14][C][logger:424]:   Level for 'template.text_sensor': ERROR
[19:13:14][C][logger:424]:   Level for 'sensor': ERROR
[19:13:14][C][logger:424]:   Level for 'binary_sensor': INFO
[19:13:14][C][logger:424]:   Level for 'cover': INFO
[19:13:14][C][logger:424]:   Level for 'switch': INFO
[19:13:14][C][logger:424]:   Level for 'climate': ERROR
[19:13:14][C][logger:424]:   Level for 'endstop': DEBUG
[19:13:14][C][logger:424]:   Level for 'text_sensor': ERROR
[19:13:14][C][logger:424]:   Level for 'homeassistant.sensor': ERROR
[19:13:14][C][sn74hc595:029]: SN74HC595:
[19:13:14][C][switch.gpio:068]: GPIO Switch 'CH1 IA'
[19:13:14][C][switch.gpio:091]:   Restore Mode: always OFF
[19:13:14][C][switch.gpio:031]:   Pin: 0 via SN74HC595
[19:13:14][C][switch.gpio:033]:   Interlocks:
[19:13:14][C][switch.gpio:037]:     CH1 IB
[19:13:14][C][switch.gpio:068]: GPIO Switch 'CH1 IB'
[19:13:14][C][switch.gpio:091]:   Restore Mode: always OFF
[19:13:14][C][switch.gpio:031]:   Pin: 1 via SN74HC595
[19:13:14][C][switch.gpio:033]:   Interlocks:
[19:13:14][C][switch.gpio:037]:     CH1 IA
[19:13:14][C][switch.gpio:068]: GPIO Switch 'CH2 IA'
[19:13:14][C][switch.gpio:091]:   Restore Mode: always OFF
[19:13:14][C][switch.gpio:031]:   Pin: 2 via SN74HC595
[19:13:14][C][switch.gpio:033]:   Interlocks:
[19:13:14][C][switch.gpio:037]:     CH2 IB
[19:13:14][C][switch.gpio:068]: GPIO Switch 'CH2 IB'
[... 210 lines cut ...]
[19:13:16][C][endstop.cover:106]:   Open Endstop 'BEMF CH3 sensor'
[19:13:16][C][endstop.cover:107]:   Open Duration: 60.0s
[19:13:16][C][endstop.cover:108]:   Close Endstop 'BEMF CH3 sensor'
[19:13:16][C][endstop.cover:109]:   Close Duration: 60.0s
[19:13:16][C][endstop.cover:105]: Endstop Cover 'CH4'
[19:13:16][C][endstop.cover:105]:   Device Class: 'shutter'
[19:13:16][C][endstop.cover:106]:   Open Endstop 'BEMF CH4 sensor'
[19:13:16][C][endstop.cover:107]:   Open Duration: 60.0s
[19:13:16][C][endstop.cover:108]:   Close Endstop 'BEMF CH4 sensor'
[19:13:16][C][endstop.cover:109]:   Close Duration: 60.0s
[19:13:16][C][endstop.cover:105]: Endstop Cover 'CH5'
[19:13:16][C][endstop.cover:105]:   Device Class: 'shutter'
[19:13:16][C][endstop.cover:106]:   Open Endstop 'BEMF CH5 sensor'
[19:13:16][C][endstop.cover:107]:   Open Duration: 60.0s
[19:13:16][C][endstop.cover:108]:   Close Endstop 'BEMF CH5 sensor'
[19:13:16][C][endstop.cover:109]:   Close Duration: 60.0s
[19:13:16][C][endstop.cover:105]: Endstop Cover 'CH6'
[19:13:16][C][endstop.cover:105]:   Device Class: 'shutter'
[19:13:16][C][endstop.cover:106]:   Open Endstop 'BEMF CH6 sensor'
[19:13:16][C][endstop.cover:107]:   Open Duration: 60.0s
[19:13:16][C][endstop.cover:108]:   Close Endstop 'BEMF CH6 sensor'
[19:13:16][C][endstop.cover:109]:   Close Duration: 60.0s
[19:14:13][D][main:166]: Calibrate CH1_cover
[19:16:43][D][endstop.cover:085]: 'CH1' - Max duration reached. Stopping cover.
```

heinemannj, 2023-10-28:
I've installed an old ESP-WROOM-32 as well - the same behavior as on ESP32-WROOM-32D.

After power on outside the motor_shield board the following behaviour:

[image]

The diff between running with the board and outside the board:

[image]

[image]

nliaudat, 2023-10-28:
You should have some
`CH1 closing endstop reached at 0.109889 V`

What are the values you setup for bemf_triggers ?

[image]

heinemannj, 2023-10-28:
I've reviewed https://github.com/nliaudat/esp32_8ch_motor_shield/issues/9

The same figures.
The only difference: I have a newer board
v57 rev 1.48

Putting a multimeter between GPIO36 and GND gives everytime **0mV**.

[image]

Putting a multimeter between below and GND gives around 20mV when the motor is running unhindered and around **53mV** at end stops

[image]

To simplify I've modified the setup as following:

```
esphome:
  name: floor-heating-1
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

api:
  reboot_timeout: 30min
  encryption:
    key: "Vs6v9r4OIlSVb4kgKrYRcDLisRDo6uy+vIYIFvuzi2E="

web_server:
  port: 80

sn74hc595:
  - id: 'sn74hc595_hub'
    data_pin: GPIO16 #Pin connected to SN74HC595 SER input
    clock_pin: GPIO5 #Pin connected to SN74HC595 SRCLK pin
    latch_pin: GPIO17 #Pin connected to SN74HC595 RCLK pin
    oe_pin: GPIO18 ### needed by esphome, but connected to GND (outputEnable is used only for pwm all th pin)
    sr_count: 2 #Number of daisy-chained shift registers, up-to 4

# Enable logging
logger:
  level: DEBUG #NONE# ERROR #INFO #DEBUG  #VERBOSE
  baud_rate: 0 #to disable logging via UART
  logs:
    adc: DEBUG

switch:
    ### CH1
  - platform: gpio
    name: "CH1 IA"
    pin:
      sn74hc595: sn74hc595_hub
      number: 0
      inverted: False
    internal: false
    id: CH1_IA_pin
    interlock: &interlock_group_CH1 [CH1_IA_pin, CH1_IB_pin]
    restore_mode: always off
  - platform: gpio
    name: "CH1 IB"
    pin:
      sn74hc595: sn74hc595_hub
      number: 1
      inverted: False
    internal: false
[... 160 lines cut ...]
[15:13:51][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:51][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:52][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:52][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:53][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:53][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:54][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:54][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:55][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:55][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:56][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:56][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:57][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:57][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:58][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:58][D][switch:016]: 'CH1 IB' Turning OFF.
[15:13:58][D][switch:055]: 'CH1 IB': Sending state OFF
[15:13:58][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:59][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:13:59][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:14:00][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:14:00][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:14:01][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
[15:14:01][D][sensor:094]: 'BEMF 1': Sending state 0.07500 V with 2 decimals of accuracy
```

nliaudat, 2023-10-28:
If you want help, do not use custom firmware as some changes may trouble : 
`update_interval: 500ms`

0.07500 V means 0V

You must give more information !

- What are the values before and after the LM324 module, 
- Values between BEMF1 and GND, 
- 3,3V real voltage as ADC is link to vref
- etc

nliaudat, 2023-10-28:
[image]

nliaudat, 2023-10-28:
Your screenshots are the same as https://github.com/nliaudat/esp32_8ch_motor_shield/discussions/41

Not an issue ! [WMP](https://github.com/WMP)

heinemannj, 2023-10-28:
Ohh - very precise instruction - Thanks:

1) without motor activity: 3.29 V 
1) with motor activity (endpoint): 3.29 V

2) without motor activity: 0.000 V
2) with motor activity (endpoint): 0.055 V

3) without motor activity: 0.000 V
3) with motor activity (endpoint): 0.053 V

4) without motor activity: 0.000 V
4) with motor activity (endpoint): 0.004 V

**5) I have the prevois board - how to measure?**

6) without motor activity: 0.000 V
6) with motor activity (endpoint): 0.000 V

Same board as in https://github.com/nliaudat/esp32_8ch_motor_shield/discussions/41
Potentially other valves

heinemannj, 2023-10-28:
For above testings I've modified config to your actual esphome settings:

```
sensor:
#ADC2 pins cannot be used when Wi-Fi is used

### new version with opamp
## in use the motor use 15-20 mV. At endstop, it raise to 50 mV
  - platform: adc
    pin: GPIO36
    name: "BEMF 1" #CH1 + CH2
    update_interval: 100ms
    attenuation : 0db 
    accuracy_decimals: 3
    filters:
    - offset: -0.075
    - sliding_window_moving_average:
        window_size: 9
        send_every: 9
    id : BEMF_1_2_sensor_ADC
```

heinemannj, 2023-10-28:
> **5) I have the prevois board - how to measure?**

[image]

5a) without motor activity: 0.000 V
5b) with motor activity (endpoint): 0.048 V

nliaudat, 2023-10-28:
GPIO36 is not ADC2 if you use default board config : 
```
esp32:
  board: nodemcu-32s
```

[image]

**Following the information you provided, I mean you LM324 opamp is not working.**

Have you jumper cable to bypass it ?

heinemannj, 2023-10-28:
GPIO36 is not ADC2 if you use default board config:
I don't understand ...

Cable bypass: 

I've tried with a simple unsoldered cable (
Flow tested) - but no progress

[image]

nliaudat, 2023-10-28:
You had good values in 2b) with motor activity (endpoint): 0.055 V. I do not understand why the opamps kills the voltage :(

1st) With GPIO36, I mean perhaps you have a different esp32 devkit board with some different pinout.
Can you check the label linked with CH1-2BEMF ?

Some references for pinout : 
https://www.upesy.com/blogs/tutorials/esp32-pinout-reference-gpio-pins-ultimate-guide?shpxid=501c40ad-44e6-487b-9aaa-15e0d69a7f22

2nd) If you can solder, perhaps the best way to go is cutting the BEMF1-2 cutout and manual wiring : 

[image]

heinemannj, 2023-10-28:
1) NO different pin layout

[image]

2) I can solder, but I'm asking myself why to disturb your nearly newest board design?

If you can support with a working hardware solution - highly appreciated.
To be frankly: For me it's not a question of money ...

I want's to have a prove of concept that an "outside temperature and weather-compensated return flow temperature heating control"  can be realized very smoothly with your board, esphome and Home Assistant.

https://github.com/nliaudat/esp32_8ch_motor_shield/issues/4
https://github.com/heinemannj/Hydraulic-Balancing/edit/main/README.md

I've already modified your code and the mayority is working well ...
I need controlable valves to confirm - and for sure I will share within your project.

nliaudat, 2023-10-28:
I really mean your LM324 opamp module is dead !

If you want to debug further, we need more steps. I have really tested all released version of the board and all are working. (The first one had a shortcut between CH7 & 8, that wasn't detected cause the valves cannot run at same time)

Do you have a laboratory power supply ? If not, a 1.5V battery will do half the job. 

1) LM324 test:
You must have 6x the input voltage as output per channel

[image]

2) Will need to test ADC inputs with direct power supply on GPIO36 (CH1-2), GPIO39 (CH3-4), GPIO34(5-6), GPIO35(7-8).
Some bad ESP32 boards do not have onboard calibration data and may have strange results.

3) Have you the same behaviour with channel 7 (or others) ?

heinemannj, 2023-10-28:
Can you please provide me a working board plus ESP32 board for further testing - I beleave hardcore debugging is out of my possibilities ... - Maybe it's to late this evening ...

SDorry

heinemannj, 2023-10-29:
Ok - Let's try to move forward.

**Actually I've a broken floor heating system - the old 6x valves (on/off) are broken and needs to be replaced.
The 6x new homematic valves are not controllable at this time.
=> all six heating circuits are open all the time, not the worstcase because my heating pump is more or less able to control ...**

For further tests I'm using a battery: 1.208 V

[image]

**LM324 test (without ESP32):**

```
For all channels 1) - 8):
Between VCC and GND: 0.888 V from battery

12in and GND: 0.572 V
12out and GND: 0.004 V

34in and GND: 0.572 V
34out and GND: 0.004 V

56in and GND: 0.572 V
56out and GND: 0.004 V

78in and GND: 0.572 V
78out and GND: 0.004 V
```

Same figures as above if I'm powering via board power supply - again without ESP.

`Between VCC and GND: 3.23 V from power supply`

If I understand you correctly above measurement are proven evidences that my board is broken - Right?

Please confirm.

nliaudat, 2023-10-29:
The opamp must have 3.3V power to work. 
Have you power on it ?

Regards

heinemannj, 2023-10-29:
**ADC inputs with direct power supply
on GPIO36 (CH1-2), GPIO39 (CH3-4), GPIO34(5-6), GPIO35(7-8) (whitout conroller board):**

[image]

```
Power value by multimeter: 1.347 V
By ESP on GPIO36: 1.294 V
```
```
  - platform: adc
    pin: GPIO36
    name: "BEMF 1" #CH1 + CH2
    update_interval: 100ms
    attenuation : auto 
    accuracy_decimals: 3
    filters:
    - offset: -0.075
    - sliding_window_moving_average:
        window_size: 9
        send_every: 9
    id : BEMF_1_2_sensor_ADC
```

=> With offset **-4% difference**
=> Without offset 1.369 V =>  **+1.6 % difference**

[image]

Shortcut between GPIO36, GPIO39, GPIO34 and GPIO35:

[image]

```
sensor:
#ADC2 pins cannot be used when Wi-Fi is used

### new version with opamp
## in use the motor use 15-20 mV. At endstop, it raise to 50 mV
  - platform: adc
    pin: GPIO36
    name: "BEMF 1" #CH1 + CH2
    update_interval: 100ms
    attenuation : auto 
    accuracy_decimals: 3
    filters:
    - offset: -0.075
    - sliding_window_moving_average:
        window_size: 9
        send_every: 9
    id : BEMF_1_2_sensor_ADC

  - platform: adc
    pin: GPIO39
    name: "BEMF 2" #CH3 + CH4
    update_interval: 100ms
    attenuation : auto 
    accuracy_decimals: 3
    filters:
    - offset: -0.075
    - sliding_window_moving_average:
        window_size: 9
        send_every: 9
    id : BEMF_3_4_sensor_ADC

  - platform: adc
    pin: GPIO34
    name: "BEMF 3" #CH5 + CH6
    update_interval: 100ms
    attenuation : auto 
    accuracy_decimals: 3
    filters:
    - offset: -0.075
    - sliding_window_moving_average:
        window_size: 9
        send_every: 9
    id : BEMF_5_6_sensor_ADC

  - platform: adc
    pin: GPIO35
    name: "BEMF 4" #CH7 + CH8
    update_interval: 100ms
    attenuation : auto 
    accuracy_decimals: 3
    filters:
    - offset: -0.075
    - sliding_window_moving_average:
        window_size: 9
        send_every: 9
    id : BEMF_7_8_sensor_ADC
```

[image]

`Power value by multimeter: 1.364 V`

=> With offset 1.310/1.364 = -4% difference
=> Without offset (1.310+0.075)/1.364 V => +1.54 % difference

heinemannj, 2023-10-29:
> The opamp must have 3.3V power to work. Have you power on it ?
> 
> Regards

Yes - this was my second round of testing with 3.3 V from the board without ESP

nliaudat, 2023-10-29:
ADC works but LM324 is dead. 

The only solution is to cut the BEMF linked to opamp

[image]

It can be reversed if you can solder ....

heinemannj, 2023-10-29:
Must I really cut the connections from BEMF to GPIOs or are the shortcuts between the GPIOs the different channels sufficient enough?

I have Pin headers and I can solder them.
 The shortcut with jumper cables are also possible.

heinemannj, 2023-10-29:
I've found a potentiometer to **reduce the input power** :-)

[image]

[image]

`Power value by multimeter: 0.073 V`

=> With offset 0.007/0.073 V = -4.1% difference
=> Without offset (0.007+0.075)/0.073 V => +12.3 % difference

**Second try after changing the ADC offset from 0.075 to 0.000 V:**

`Power value by multimeter: 0.068 V`

[image]

=> Without offset 0.076)/0.068 V => +11.8 % difference

`=> The lowest power I can measure with my ESP32-WROM-32D is: 68 mV`

**With a maximum sensitivity of 68 mV, my ESP cannot detect motor activity and especially the endpoint (measured as ~55 mV).**

Question:
Can a self made calibration of the ESP32 can shift up the ADS sensitivity?

nliaudat, 2023-10-29:
They are too many problems ! 
You ESP32 board do not read ADC under 68mv (as it should read 20mv to 3.3v)
The opamp do not work

There are 2 solutions : change either shield or ESP32.

heinemannj, 2023-10-29:
OK - I will change both.

What kind of ESP do you recommend and where to buy?
From whom I can order a new board?

nliaudat, 2023-10-29:
Some people sells boards in https://github.com/nliaudat/esp32_8ch_motor_shield/discussions
You can order yourself at jlcpcb.com

heinemannj, 2023-10-29:
Are there better alternatives than ESP32-WROOM-32D which can be used by the board?

nliaudat, 2023-10-29:
new ESP32 C6 or S6 have better ADC. I haven't tested actually.

heinemannj, 2023-10-29:
They are really fitting into the board sockets - I beleave they have 2 or four pins less?

heinemannj, 2023-10-29:
I try to order boards at jlcpcb.com, but I get the following error message:

[image]

heinemannj, 2023-10-30:
I've ordered at jlcpcb.com

In the meantime I will try the approach to dual ads1115 as adc for all channels as proposed by https://github.com/nliaudat/esp32_8ch_motor_shield/discussions/19#discussioncomment-3724593

heinemannj, 2023-10-30:
[image]

nliaudat, 2023-10-30:
> I try to order boards at jlcpcb.com, but I get the following error message:
> 
> [image]

Normally, that's all holes or pin headers

WMP, 2023-10-31:
My discusion : https://github.com/nliaudat/esp32_8ch_motor_shield/discussions/41 and this issue describe the same problem. I gave my PCB with the actuators and esp32 to an electrical engineering service. After some time I hear from the electrical engineer that during start-up the actuators have a very high current, several A's. The electrical engineer added a 200 µF capacitor but it didn't help. Today I got the information (fast, no specifics) that the problem is a badly sized LM324, because the actuators have a much higher short-circuit current than the LM324 can handle. The engineer is to find a better model.

Checking the specification, I see that all LMs.... have an Isc of 40mA to 60mA, and the actuator in the first picture says max 120mA. In the LM specification it says: "Short circuits from outputs to VCC can cause excessive heating and eventual destruction." . So you may be right in saying that the LM324 is faulty, as it could have been damaged by the first actuator start. 

I wonder if the use of LM is even possible for these actuators.

heinemannj, 2023-10-31:
After hardcore tests porposed during last weekend my board related to BEMF
ist completly dead - direct shortcut on all 8 BEMF channels ...

This is may be inline with above findings ....

Cheers
Joerg

nliaudat, 2023-11-01:
The LM324 module capture the value of BEMF ([Back electrical motor force](https://www.precisionmicrodrives.com/ab-021)).
I found very strange that the BEMF would be greater than the max current handled by motor (120ma)
I'll make tests.
@Karl-opec : What do you think of @WMP comments about the LM324 ?

Regards

heinemannj, 2023-11-04:
I have ordered and tested a third ESP32D:

`=> The lowest power I can measure with my ESP32-WROM-32D is: 75 mV`

ESPHome documentation is clearly stating, that on ESP32 the lowest adc voltage is ~0.075V:
https://esphome.io/components/sensor/adc.html

[image]

heinemannj, 2023-11-04:
Testing the approach with a single ads1115:

`=> The lowest power I can measure with my ESP32-WROM-32D is: 0.000040 V => 0.04 mV`

[image]

The usage of a single ads1115 is providing:

- a simple solution which is undependant from ESP32 adc capabilities
- sensitive and reliable measurements

heinemannj, 2023-11-04:
Setup:

[image]

nliaudat, 2023-11-04:
ads1115 is like LM324 but much more expensive

Karl-opec, 2024-02-24:
> The LM324 module capture the value of BEMF ([Back electrical motor force](https://www.precisionmicrodrives.com/ab-021)). I found very strange that the BEMF would be greater than the max current handled by motor (120ma) I'll make tests.
> 
> The Isc of 40mA to 60mA is the "max output current" the LM324 can boosts per channel, not related about input.
> 
> @Karl-opec : What do you think of @WMP comments about the LM324 ?
> 
> Regards

The LM324 does not drive the valves. it only amplifies the voltages  over current resistors to get them in workable ranges  for measurement by the esp32. the blocking current of the valves are approx. 35mA. Over the resistor this gives 35mV. The LM324 amplifies these voltages by 6 to approx. 180mV. This is the voltage at which the ESP32 switches of the valve.

Karl-opec, 2024-03-15:
Issue #55 is fits to this problem also.

WMP, 2024-03-15:
Yes, it would be great if a fixed version with rs11 was published.

closed 2024-03-15 by nliaudat