nliaudat/esp32_8ch_motor_shield #50: Need help - Order Error jlcpcb.com

grericht, 2024-02-20:
Get this Error ordering by jlcpcb.com

The below parts won't be assembled due to data missing.
WROOM-LEFT1,WROOM-RIGHT,WROOM-LEFT,HLK-PM03,P1,P2,P3,P4,H4,H5,OPAMP-H1,U1 designators don't exist in the BOM file.

[image]

nliaudat, 2024-02-20:
For P5 to P12 +  Screw-230V, select part #C8465
H1, H3 are holes, the is no part

nliaudat, 2024-02-20:
[BOM_esp32-8ch-dc-motor-shield_AC_2024-02-20.csv](https://github.com/nliaudat/esp32_8ch_motor_shield/files/14349458/BOM_esp32-8ch-dc-motor-shield_AC_2024-02-20.csv)
New BOM file

nliaudat, 2024-02-20:
https://github.com/nliaudat/esp32_8ch_motor_shield/releases/tag/v58_rev_1.52

closed 2024-02-20 by nliaudat

grericht, 2024-02-23:
new bom file still not working. but using part #C8465 helps!