nliaudat/esp32_8ch_motor_shield #56: Need help

rzerres, 2024-05-10:
Try to improve the ordering via JLPCB

Dear Nicolas,

This is great work.

Since I'm rebuilding an apartment, I'd like to use this PCB to control my newly installed underfloor heating.
I was amazed and completely satisfied when i got hold of this repository. I do have 4 circles that should be managed via individual porportional actors.
The given documentation my be a bit rough for newcomers like me. So i took the time to prepare a PR that introduces an "HOWTO_get_going.md".
I will submit it once i finished some last shaping.

May you can enlight me to solve following questions:

1. Is the given box design (box-V57rev-1.48) compatible with the upstream controller (PCB_PCB v58 rev 1.51_2023-10-02.pdf)?
2. The controller is fully configurable via home-assistant?
3. I'd like to order via JLPCB
   Since it will be the first time usage for me, I'd like to describe the process to ease the process.
   - register an accout
   - which "part list" has to be uploaded in the JLPCB Gui to include
     * ESP32-WROOM-32D or ESP32-WROOM-32U
	 * The Controller is able trigger the motors of open proportional 
	   (like percentage, triggered by temprature sensor?
 4. Can you advise a good input AC Power (presumable a 5V for HKL-5MX)?

nliaudat, 2024-05-10:
Hi, 

I'll try to respond quickly : 

1) no, the box is for RJ11 version. It may fit, but I suggest you to investigate the openscad version : [PCB_box_generator_v1-1.scad](https://github.com/nliaudat/esp32_8ch_motor_shield/blob/main/box/general/PCB_box_generator_v1-1.scad)
2) Yes, the controller is fully compatible with hassio as it runs esphome.
3) 

- a : 

ESP32-WROOM-32D or U version (with external antenna) has to be bought elsewhere than JLCPCB cause the price in JLCPCB is huge. You may find your needs in aliexpress or others. 

- b: 

The controller works in proportionnal mode. The temperature difference between target temp and real temp triggers a movement every 15 minutes. There is a version of firmware with a PID controll.  You can control it with hassio or standalone. You'll find futher informations on https://github.com/nliaudat/floor-heating-controller/wiki/

4) For the Ac power, most proportionnal valves use 3.3V as esp32. I'll recommend to use HKL-5M03 or HKL-PM03
Actually, if you buy the HKL-XX component at JLCPC, the weight is misscalculated and the freight cost is huge. I'll recommend you to buy it separatly and solder yourself. 

Have a look at https://github.com/Lenti84/VdMot_Controller before buying one of mine. You may have the choice of 2 solutions. 

Regards

rzerres, 2024-05-10:
Thanks. This advise is quite helpful.

One more question:
Vendor documentation states ESP32-WROOM-32D and ESP32-WROOM-32U deprecated. If I did read correctly, ESP32WROOM32E and ESP32WROOM32UE are their successors. Will the firmware will work without adaption?

nliaudat, 2024-05-11:
> Thanks. This advise is quite helpful.
> 
> One more question: Vendor documentation states ESP32-WROOM-32D and ESP32-WROOM-32U deprecated. If I did read correctly, ESP32WROOM32E and ESP32WROOM32UE are their successors. Will the firmware will work without adaption?

All boards supported by platformio and esp-idf frameworks should compile esphome. 

https://docs.platformio.org/en/latest/platforms/espressif32.html#boards

rzerres, 2024-05-13:
How about this one?

https://www.reichelt.de/entwicklungsboard-esp32-wroom-32e-esp32devkitc32e-p341303.html?&nbc=1

nliaudat, 2024-05-14:
Hi, 

I'm surprised at the price of the card at Reichelt, which is over €10...
I would still advise you to stick with a well-known card (ESP32-WROOM-32D 38p) available on Aliexpress for example. You'll find one for 3€.

With the Gerber files, you can order a PCB from anywhere. 
I don't give the EasyEda sources any more, because some people made modifications, ordered and then complained that it didn't work. 

What's more, no one has proposed any public changes to the PCB board in 5 years.

I also no longer support older versions because some components are no longer available. 
You must use the latest version:  https://github.com/nliaudat/esp32_8ch_motor_shield/releases/tag/v58_rev_1.52
Unzip it and you'll find the files you need to ordrer.

Regards

rzerres, 2024-05-14:
Thanks. I will go with that.

And i completely understand Not offering the gerbers any more. Quite offending If someone changes the layout then complains and requires your Support. Not in line with Open source  Code of conduct.

nliaudat, 2024-05-14:
> Thanks. I will go with that.
> 
> And i completely understand Not offering the gerbers any more. Quite offending If someone changes the layout then complains and requires your Support. Not in line with Open source Code of conduct.

I offer the gerbers ! The schematics and you need. But not the EasyEda source : 
 https://github.com/nliaudat/esp32_8ch_motor_shield/releases/tag/v58_rev_1.52

closed 2024-05-22 by nliaudat